sifive/duh #56: Create a Python library for working with DUH?

mithro, 2019-09-28:
Many HDLs are written in Python (see http://bit.ly/pyfgpalibs for example). It would be awesome to have a good Python library for working with DUH.

mithro, 2019-09-28:
FYI -- There are a couple of Python libraries for IPXACT
 * pyIPXACT - https://github.com/Paebbels/pyIPXACT
 * ipyxact - https://github.com/olofk/ipyxact

mithro, 2019-09-28:
A good python library for working with DUH is probably required for FuseSoC integration?

olofk, 2019-11-21:
I agree that a Python library would be very good to have. Since it's pretty much a 1 to 1 mapping with IP-XACT it would probably not be too much work to add import/export to ipyxact. I'll look into that when I can find some time

For use with FuseSoC it's actually not required. The FuseSoC generator interface is language agnostic and only requires the generator to read and write a yaml file

drom, 2019-11-21:
There are several Python libraries using DUH. It would be easy to take https://github.com/sifive/duhportinf as an example. We have DUH <-> IP-XACT translation project and I don't see a need to duplicate it in Python just for sake of having it in Python.