sifive/duh #59: duh validate doesn't find custom buses

olofk, 2019-11-21:
I'm getting `has unknown busType` when I run duh validate on my component. The bus defintion lives in a separate directory, and I was able to get duh-portinf to use it with the `-b` flag but I can't make `duh validate` find it

drom, 2019-11-21:
Yes, @olofk most of the tools right now only works with the bus definitions from https://github.com/sifive/duh-bus . I agree that we need a good mechanism to include bus definitions from other sources. And not only bus definitions but components and designs as well.
I was thinking of NPM based solution when every DUH library would be an NPM package, exporting index of DUH documents. NPM already knows how to find, install and manage such packages from many different sources: https://docs.npmjs.com/files/package.json#urls-as-dependencies
What do you think?

adeelliaquat-lm, 2020-11-12:
Hi,
Recently we started working on Duh, and we did a similar thing. We added the support of AXI5-Lite (AMBA-5) in Duh. It is yet to be functionally verified, but we were able to run the complete Duh flow along with Duh validation and scala export. We did some modifications manually. The screenshot of the exported scala chunk is attached below: 

[image]

> Yes, @olofk most of the tools right now only works with the bus definitions from https://github.com/sifive/duh-bus . I agree that we need a good mechanism to include bus definitions from other [...]