sifive/duh #66: duh validate prints out unnecessary validation errors when DUH document not passed in

richardxia, 2020-02-07:
If you run `duh validate` with no arguments, one would normally expect to see only the usage string. Instead, you get the usage string and a number of validation errors:

```
$ npx duh validate
duh validate [filename]

validate document

Positionals:
  filename  document file                                               [string]

Options:
  --version      Show version number                                   [boolean]
  --verbose, -v                                                 [default: false]
  --help         Show help                                             [boolean]

[ { keyword: 'required',
    dataPath: '',
    schemaPath: '#/oneOf/0/required',
    params: { missingProperty: 'catalog' },
    message: 'should have required property \'catalog\'' },
  { keyword: 'required',
    dataPath: '',
    schemaPath: '#/oneOf/1/required',
    params: { missingProperty: 'component' },
    message: 'should have required property \'component\'' },
  { keyword: 'required',
    dataPath: '',
    schemaPath: '#/oneOf/2/required',
    params: { missingProperty: 'design' },
    message: 'should have required property \'design\'' },
  { keyword: 'required',
    dataPath: '',
    schemaPath: '#/oneOf/3/required',
    params: { missingProperty: 'abstractionDefinition' },
    message: 'should have required property \'abstractionDefinition\'' },
  { keyword: 'required',
    dataPath: '',
    schemaPath: '#/oneOf/4/required',
    params: { missingProperty: 'busDefinition' },
    message: 'should have required property \'busDefinition\'' },
  { keyword: 'oneOf',
    dataPath: '',
    schemaPath: '#/oneOf',
    params: { passingSchemas: null },
    message: 'should match exactly one schema in oneOf' } ]
```

This is on DUH 1.16.3.

Ramlakshmi3733, 2020-10-05:
adding an issue to it, where if the file does not exist (though mentioned), it does not give a crisp error message.
Issue in duh-core/lib/read-duh.js

Ramlakshmi3733, 2020-10-06:
waiting for duh-core pull request to be merged before submitting a pull request on duh.
But the fix has been tested.

commit 7b57f286 (2020-10-08): Merge pull request #82 from sifive/issue66

fixes #66
commit ce1a2215 (2020-10-08): fixes #66
closed 2020-10-08 by drom