sifive/duh #68: duh-bus not properly installed ERROR while running command duh-portinf

ishitapvips, 2020-02-18:
Hi,
I am trying to run duh-portinf command but it is giving the following error
<pre>Traceback (most recent call last):
  File "/home/ishita/node_modules/.bin/duh-portinf", line 16, in <module>
    duhportinf.main_portinf.main()
  File "/home/ishita/node_modules/duhportinf/pylib/duhportinf/main_portinf.py", line 202, in main
    "duh-bus not properly installed, please use -b/--duh-bus"
AssertionError: duh-bus not properly installed, please use -b/--duh-bus
</pre>

the command was running fine but after installing the newest version of duh-1.16.3 the command is giving above error

drom, 2020-02-18:
Please try again.
I have fixed `duh-bus` v0.5.5 https://github.com/sifive/duh-bus/commit/99e78af2986c607a20018ea56da41574d32c5bb3

ishitapvips, 2020-02-19:
Hi, 
I tried with the updated version but it stills gives the following error.
<pre>
ishita@terminal:/work/ishita/workspace/block-pio-sifive$ duh-portinf pio.json5 -o pio.json5
loading 11 bus specs
  - done, loaded 22 bus defs with 194 required and 186 optional ports 
mapping 0 unassigned ports
initial bus pairing with port groups
  - done
bus mapping
Traceback (most recent call last):
  File "/home/ishita/node_modules/.bin/duh-portinf", line 16, in <module>
    duhportinf.main_portinf.main()
  File "/home/ishita/node_modules/duhportinf/pylib/duhportinf/main_portinf.py", line 213, in main
    i_bus_mappings = get_bus_matches(unassn_ports, bus_defs)
  File "/home/ishita/node_modules/duhportinf/pylib/duhportinf/main_portinf.py", line 152, in get_bus_matches
    opt_i_bus_mappings = _get_initial_bus_matches(bt, opt_i_bus_pairings)
  File "/home/ishita/node_modules/duhportinf/pylib/duhportinf/main_portinf.py", line 115, in _get_initial_bus_matches
    util.progress_bar(pcurr, ptot, length=plen)
  File "/home/ishita/node_modules/duhportinf/pylib/duhportinf/util.py", line 193, in progress_bar
    percent = ("{0:." + str(decimals) + "f}").format(100 * (iteration / float(total)))
ZeroDivisionError: float division by zero
</pre>

duh --version I am using 1.16.4.

Ramlakshmi3733, 2020-10-07:
i am able to reproduce the error