sifive/duh #78: duh validate should disallow duplicate address block names

richardxia, 2020-06-29:
`duh validate` currently allows a component to have two address blocks with the same name, even within the same memory map. This example DUH document passes validation for me:

```json5
{
  "component": {
    "vendor": "my_vendor",
    "library": "my_library",
    "name": "my_block",
    "version": "0.1.0",
    "memoryMaps": [
      {
        "name": "my_memory_map",
        "addressBlocks": [
          {
            "name": "my_address_block",
            "baseAddress": 0,
            "range": 0x400,
            "width": 32,
          },
          {
            "name": "my_address_block",
            "baseAddress": 0x1000,
            "range": 0x400,
            "width": 32,
          },
        ]
      }
    ]
  }
}
```

I don't think this is valid, and I believe that IP-XACT does not allow for two address blocks within the same memory map to have the same name. Can we update `duh validate` to disallow this?

I am running duh 1.17.10, which appears to be the latest version of this repo.

drom, 2020-06-29:
related to https://github.com/sifive/duh-core/issues/9

drom, 2020-07-02:
should be fixed in v1.18.0

closed 2020-07-02 by drom