sifive/duh #80: Default value of access/modifiedWriteValue/readAction properties

taichi-ishitani, 2020-08-08:
I'm Taichi Ishitani and author of [RgGen](https://github.com/rggen/rggen).
I'm developing a plugin for DUH because @drom requested me (rggen/rggen-duh#1).

Can you tell me what is default value of `access`, `modifiedWriteValue` and `readAction` properties?
I think DUH is inspired by IP-XACT so default value of these properties should be based on IP-XACT specification.

drom, 2020-08-08:
Yes, DUH is based on IPXACT in Spirit. And Should follow IPXACT rules where it makes sense.

`access`
* if not defined on field --> inherited from parent register.
* if not defined on register --> inhereted from parent registerFile/addressBlock
* if not defined on registerFile/addressBlock --> `read-write`

`modifiedWriteValue`, `readAction` -- if not specified assume no action extra action.

taichi-ishitani, 2020-08-09:
Thank you!
I understood.

closed 2020-08-09 by taichi-ishitani