sifive/duh #81: du validate doesn't work with absolute paths.

rpadler, 2020-09-03:
when using absolute paths with duh validate, duh validate acts as if the file does not exist.

Ramlakshmi3733, 2020-10-02:
Usage for absolute paths should be:
<path>/duh.js validate file://<unix absolute path>

The issue is in duh-core/lib/read-duh.js with jsonRefs.resolveRefs({$ref: fileName}, {
it is built with relative paths and adds the `cwd` to the absolute path unless the filename is given in the file:// syntax
While it seems trivial to add the file:// to an absolute path and force it to work, concerned about any side effects in other usages.
https://github.com/whitlockjc/json-refs/issues/122

Consequently the resolution I am proposing is that user enters the absolute path with file:// <absolute path> syntax.
Relative paths do not need any changes: ./<relative path> will work.
I agree it does create a difference in entry.  

Am closing the issue with the comment.  We will be enhancing the code in the future and should make it even cleaner.

closed 2020-10-02 by Ramlakshmi3733

rpadler, 2020-10-02:
Please provide a real fix – I don’t find it acceptable to have to enter
file://

*From:* Ram Sunder <notifications@github.com>
*Sent:* Friday, October 2, 2020 12:36 PM
*To:* sifive/duh <duh@noreply.github.com>
*Cc:* Robbie Adler <robbie.adler@sifive.com>; Author <
author@noreply.github.com>
*Subject:* Re: [sifive/duh] duh validate doesn't work with absolute paths.
(#81)

Closed #81 <https://github.com/sifive/duh/issues/81>.

drom, 2020-10-02:
The tool is taking URI not a File path. This way you can validate files over other access protocols like FTP, HTTP
```
duh val https://sifive.github.io/block-ark/i2c.json5
```