sifive/duh #84: Duh Validate doesn't work for I/Os without Prefix

adeelliaquat-lm, 2020-11-04:
Can DUH be used for IPs without the prefix in its IOs? Is prefix necessary with IO ports ? 
While working on an IP without the prefix in its ports, we encountered the following the error "DUH document does not match schema"
Picture of error is attached for reference.

Thanks

[image]

drom, 2020-11-04:
Please provide more details on your error case.
By looking at the screenshot, the first `busInterface`  inside your `component` has the `name` field is too short.

adeelliaquat-lm, 2020-11-05:
Thanks for the response, let me quote an example to clarify my point.
Duh will work perfectly fine for the following chunk of interface:
case 1:
input                            t_ctrl_awvalid,
output logic                 t_ctrl_awready,
input  [addrWidth-1:0] t_ctrl_awaddr,
input                    [2:0] t_ctrl_awprot,

but will not work (will not validate) for the following chunk of interface:
case 2:
input                            awvalid,
output logic                 awready,
input  [addrWidth-1:0] awaddr,
input                    [2:0] awprot

Moreover, i'v attached the names of the bus interfaces. It shows the name of the first bus interface that you are talking about, please have a look at it. 
It is important to note, that if we had interface like case 1 there would be a prefix written after "prefix_" and we wouldn't be having any error. The prefix in case 2 would be t_ctrl_. 

[image]

> Please provide more details on your error case.
> By looking at the screenshot, the first `busInterface` inside your `component` has the `name` field is too short.

drom, 2020-11-24:
Is this an issue with `duh-portinf` command? that produces a busInterface with empty name?