sifive/duh #85: Design option in "Block Type" doesn't create any  (.json5) base document

adeelliaquat-lm, 2020-11-04:
When we run the duh init command for creating a base document and select "Design" in "Block type", it doesn't create anything. The flow works perfectly fine for component, but not for design. Is the design part still under process ? If yes, then what additional value will it add to the Duh document ? 

Thanks

drom, 2020-11-04:
yes, `init` command needs to be fixed. PR is welcome.

adeelliaquat-lm, 2020-11-09:
> yes, `init` command needs to be fixed. PR is welcome.

Thanks for the response, Can you add some more details about the intent of the 'Design' option? I have gone through the basic description given in the below-mentioned link, but couldn't understand much.

Link: https://github.com/sifive/duh/blob/master/docs/design.md